bphermansson/EspSparsnasGateway #1: Port or merge this project to the existing OpenMQTTGateway project?

Hedda, 2018-01-23:
@bphermansson might it be a good idea to port or merge this to the existing OpenMQTTGateway project and maybe join their development team?

https://github.com/1technophile/OpenMQTTGateway

OpenMQTTGateway is also a ESP8266 and RFM69 based RF (and IR) to MQTT project however it also keeps compatibility with ESP32 and Arduino + W5100 too, and as an additional bonus it supports InfraRed transmitters and recievers as well:

https://1technophile.github.io/OpenMQTTGateway/

Blog:

https://1technophile.blogspot.se/

RM69 specific blog:

https://1technophile.blogspot.se/2017/09/why-adding-rfm69-compatibility-to.html

Wiki:

https://github.com/1technophile/OpenMQTTGateway/wiki

Good or bad idea?

Hedda, 2018-01-24:
@bphermansson Florian (a.k.a. @1technophile ) wrote here that should be possible if you are willing:

https://gitter.im/OpenMQTTGateway/Questions_support

OpenMQTTGateway project needs a developer who owns a Sparnas and has own interest to add it.

bphermansson, 2018-01-24:
Interresting! Seems like a nice project in many ways. I'll take a look and see what can be done.

Hedda, 2018-01-25:
@bphermansson  Cool! and FYI; another somewhat similar project using ESP8266 and supporting RFM69 is "MySensors":

https://github.com/mysensors/MySensors
https://www.mysensors.org

Interestingly, MySensors (which is a mature project) also happens to come from Swedish developers!

https://github.com/mysensors/MySensors/issues/1034

Posted another seperate feature request for them to implement Sparnas to as linked above.

dogmatic69, 2019-01-21:
This should be closed, it has been a year and these other projects are not interested.

MySensors closed the ticket
OpenMQTTGateway says if they wanted a meter they would build the led blink reader and skip sparsnas.

It's a dead project before it even starts. Had @Hedda perhaps make PR's to these other projects it might have been different but I doubt anything is going to happen.

closed 2019-01-22 by dogmatic69