bphermansson/EspSparsnasGateway #8: Setting the frequency via Mqtt

bphermansson, 2018-02-12:
Let the code listen for incoming Mqtt messages with a setting for the receiver frequency. To set the frequency, send something like this:
mosquitto_pub -h 192.168.1.79 -t 'EspSparsnasGateway/settings' -m '{freq:867.999939}'

fredrike, 2018-02-12:
Is it possible to set the sensor ID too?

bphermansson, 2018-02-12:
That would be nice, will give it a try.

bphermansson, 2018-02-18:
The device can now be controlled via Mqtt. We can set the frequency and the senderid, the settings are stored in the Eeprom. We also can clear the settings and reset the device.
But I think theres a bug when setting the frequency, it gets it wrong somehow?

Linuxsmurfen, 2018-04-07:
Could it be that it sets the "ifreq = 868100000;"  on row 317 ?
I tried to remark it and now i'm able to set the frequency via mqtt.

bphermansson, 2018-04-14:
I think that row was for testing. Can you set the frequency and get it right? I haven't managed to get the value set, ie if I set it to 868100000 its reports something like 868000256 back as "Listening on ...".

Naesstrom, 2018-12-10:
I commented out the line on 317 to that @Linuxsmurfen mentioned above, using `EspSparsnasGateway/settings/frequency -m 867.99` give me
```
Message arrived: EspSparsnasGateway/settings/frequency
Payload: 867.99
Frequency change
Set frequency to: 867.99
Wrote: 8
Wrote: 6
Wrote: 7
Wrote: .
Wrote: 9
Wrote: 9
```
Then it reboots and ends the logging with:
```
Stored data: 
Found a stored frequency
Stored frequency: 867.99
Calculated frequency: 867990000
There are no stored senderid, using default value
Senderid: 570631
Frequency: 867990000
61.04
Over The Air programming enabled, port: EspSparsnasGateway
Ready
IP address: 192.168.1.201
EspSparsnasGateway @ IP:192.168.1.201 SSID: Katrinelund
In initialize, frequency = 867990000
Adjusted freq: 14221148
61.04
RFM69 init done
Radio initialized.
Listening on 1023999936hz. Done in setup.
```
So even if it says it's saved the new data and repeats it as stored it ends with listenin on another frequency?

bphermansson, 2018-12-19:
Yes. Haven't managed to figure out why.

whyz, 2019-01-09:
fwiw, I have the exact same issue - the requested frequency isn't set. Happy to help in any way I can.

kalj, 2019-01-16:
I guess we can have another go at this now that we have figured out the problem with #20 & #22 ?

dogmatic69, 2019-01-20:
Setting frequency seems to now work, as long as you remove the other line as mentioned above. 

Listening on 102399999936hz seems to be a problem with wiring, maybe time to close this ticket?

closed 2019-01-22 by dogmatic69