bphermansson/EspSparsnasGateway #9: Calculates wrongly when low/none power usage

tubalainen, 2018-02-12:
The code calculates watts extremely oddly when there isnt a significant load, for example, a load greater than 1000 W.

It reports like 39302 watts when there should be 7 W only.

bphermansson, 2018-02-15:
This we have to look into. Too bad I only have one sender and it's mounted at the power meter. Will see if it's possible to fake values in the code to trigger this bug.

Sperra, 2018-02-18:
Even I have problems with incorrect and extreme values ​​of effects up to 1.6MW on a 20A main circuit.

The power is calculated according to "watt = (float) ((3600000 / PULSES_PER_KWH) * 1024) / (power);"
Here in the code "power" takes 2bytes.

A high value of "power" will give a low power in W, a value of 65535 produces a minimum power of about 56W. Lower effects seem to get the value of "power" to switch over and start from 0. This seems to match my OpenHab which shows 1.572MW power output. 1.572MW should then correspond to just under 56W the wattage is more reasonable when the values ​​around such a peak are around 80W.

bphermansson, 2018-02-19:
I'm thinking of letting the code just send raw values, calculations then has to be done by the receiver. 

Wonder if this error also shows with Strigeus code, https://github.com/strigeus/sparsnas_decoder?

tubalainen, 2018-02-25:
Looks like the bug exists in the original code as well. Or, there is something fishy with my sparsnäs used for that measurement (got two here at home). The display shows correct values, but both SDR and ESP solution shows the same..

One "ugly fix" is to consider everything under 100 W as 0 in the code... That would help me.

bphermansson, 2018-02-25:
Hmm wait... power is stored in the code as an int, and can therefor be a signed number from -32768 to 32767. The value in the data from the sender is 16 bits with a max of 65535. Not good.
Changing the power type to uint should fix this issue.

bphermansson, 2018-02-25:
That fix is in the code now, hope it works.

fredrike, 2018-02-26:
According to @tubalainen did the `u_int` not help. I've been talking to @bhaap today and apparently was this issue introduced on @tubalainen's code when I changed `PULSES_PER_KWH` from a defined constant to an `int`.

```c
        float watt =  (float)((3600000 / PULSES_PER_KWH) * 1024) / (power);
```

Do we have any samples of output with what the `power` is after the change to `u_int`?  Would it be better to have `power` as `u_int16_t`?

PR #13 (2018-03-03): fix for power calculation when data4 is 0
Apparently is data4 set to 0 when the power usage is low (around <400W in 10'000 pulses per kwh) and then is watt calculated as `power * 0.24 / PULSES_PER_KWH`.

fixes #9
closed 2018-03-03 by bphermansson