bphermansson/EspSparsnasGateway #19: 'StaticJsonBuffer' was not declared in this scope

bphermansson, 2018-07-06:
The code fails to compile if you updated ArduinoJson recently. This is because the Library Manager installs version 6.0 beta of ArduinoJson. The solution is to downgrade to the stable version 5.13.2. 
See https://github.com/bblanchon/ArduinoJson/issues/756 for more info.

bblanchon, 2018-07-09:
You can verify that the right version of ArduinoJson is installed by checking the [ARDUINOJSON_VERSION](https://arduinojson.org/v5/api/utils/version/?utm_source=github&utm_medium=issues) macros.

Netoperz, 2018-09-05:
Yap, i can confirm that bug.
all beta 's have problem, and only 5 works fine, , question is why arduino is pushing the updates to beta versions? should't that be only production version availble and dev only for those who want that ? :D

dogmatic69, 2019-01-16:
You choose the version you want when you install it, so if you don't want beta don't choose it.

PR #32 (2019-01-22): Update requirements and include the wiring diagram in the readme
fixes #19
closed 2019-01-22 by dogmatic69