bphermansson/EspSparsnasGateway #20: Listening to the wrong frequency

Freland, 2018-08-05:
For some reason the log gives me "Listening on 1023999936hz. Done in setup." even though I've set the frequency: "Stored frequency: 867999900". Since it is listening to the wrong freq. I don't get any data. I've tried updating the freq. with MQTT with no luck.

Logfile:
```
Debug on
Vcc=3014
Stored data: 1
There are stored settings
Stored frequency: 867999900
Stored sender id: d2a94a
Over The Air programming enabled, port: EspSparsnasGateway
Attempting MQTT connection...Connected to Mqtt broker as EspSparsnasGateway
Ready
IP address: 192.168.1.XX
EspSparsnasGateway @ IP:192.168.1.XX SSID: XXXXXX
Stored freq: 867999900
RFM69 init done
Radio initialized.
Listening on 1023999936hz. Done in setup.
```

kalj, 2019-01-13:
See #22 and #25 . Can probably be closed.

dogmatic69, 2019-01-18:
I have been debugging this tonight and think I have found the problem. The frequency is not written to the registry on the board (or possibly the read is not working)

```
  Serial.print("frequency: ");
  Serial.println(frequency);
  for (uint8_t i = 0; CONFIG[i][0] != 255; i++) {
    Serial.print("writeReg: ");
    Serial.print(CONFIG[i][0]);
    Serial.print(": ");
    Serial.println(CONFIG[i][1]);
    writeReg(CONFIG[i][0], CONFIG[i][1]);
    yield();
  }
```
gives the following output (trimmed for brevity):

```
22:35:08.830 -> frequency: 1422114
....
22:35:08.830 -> writeReg: 7: 21
22:35:08.830 -> writeReg: 8: 179
22:35:08.830 -> writeReg: 9: 34
```

Whilst the `getFrequency` method call has this:

```
  Serial.println("readReg: ");

  Serial.print(REG_FRFMSB);
  Serial.print(": ");
  Serial.println((readReg(REG_FRFMSB)));

  Serial.print(REG_FRFMID);
  Serial.print(": ");
  Serial.println(readReg(REG_FRFMID));

  Serial.print(REG_FRFLSB);
  Serial.print(": ");
  Serial.println(readReg(REG_FRFLSB));

  Serial.print("frequency: ");
  Serial.println((
    ((uint32_t)readReg(REG_FRFMSB) << 16) +
    ((uint16_t)readReg(REG_FRFMID) << 8) +
    (uint8_t)readReg(REG_FRFLSB)
  ));
```

And the output is: 
```
22:35:08.863 -> readReg: 
22:35:08.863 -> 7: 255
22:35:08.863 -> 8: 255
22:35:08.863 -> 9: 255
22:35:08.863 -> frequency: 16777215
```

kalj, 2019-01-19:
@dogmatic69 Are your issue the same as the one we eventually sorted out in #22 ? Many of us had this exact problem when the MOSI/MISO wires were swapped and/or leaving the RST pin of the Adafruit RFM69HCW BoB unwired to ground. Using the new schematic #27 it seems we all get it to work. Also, have you included the error checks in #25? Those can help ruling out a failed initialization.

dogmatic69, 2019-01-19:
Was missing the reset connected to ground! Now the registries are written.

bphermansson, 2019-01-19:
Great work! Since I lack time for this project I invited Naesstrom and Freland as collaborators. Anyone else wanna join?

kalj, 2019-01-19:
@bphermansson I would love to tag along. :)

dogmatic69, 2019-01-23:
This should now be sorted. If still getting that it listens to the wrong frequency check the wiring and make sure that registries that are written to actually have the correct values (eg: write -> read -> compare value). If that does not work the frequency is not being written to the registry and is probably due to wiring issue.

closed 2019-01-23 by dogmatic69