bphermansson/EspSparsnasGateway #22: No values?

Naesstrom, 2018-10-31:
So I finally got around to assembling this, even added a sma contact and a 868 antenna to the breakout board.
Serial monitor just gives me this:
```
Attempting MQTT connection...Connected to Mqtt broker as EspSparsnasGateway
Stored data: 
There are no stored frequency, using default value
There are no stored senderid, using default value
Senderid: 570631
Frequency: 868000000
61.04
Over The Air programming enabled, port: EspSparsnasGateway
Ready
IP addre
```
The senderid is the one I entered and the frequency looks right to, the strange thing is that it cuts out before it even tells me the IP... Debug is set to 1 btw.

On the mqtt side of things it repeats the following 4 rows a couple of times
```
{"status":"Connected to Mqtt broker as EspSparsnasGateway"}
{"status":"There are no stored frequency, using default value"}
{"status":"There are no stored senderid, using default value"}
{"status":"EspSparsnasGateway @ IP:192.168.1.201 SSID: Katrinelund"}
```

The really strange thing tough is that if I disconnect the ESP from everything and run it the monitor seems to work?
```
Attempting MQTT connection...Connected to Mqtt broker as EspSparsnasGateway
Stored data: 
There are no stored frequency, using default value
There are no stored senderid, using default value
Senderid: 570631
Frequency: 868000000
61.04
Over The Air programming enabled, port: EspSparsnasGateway
Ready
IP address: 192.168.1.201
EspSparsnasGateway @ IP:192.168.1.201 SSID: Katrinelund
In initialize, frequency = 868100000
Adjusted freq: 14222950
61.04
RFM69 init done
Radio initialized.
Listening on 1023999936hz. Done in setup.
```

Naesstrom, 2018-10-31:
Did some more debugging, running the ESP with everything but the RFM69 unit I get a complete startup in the serial monitor to... starting to think that my module might be DOA...

bphermansson, 2018-10-31:
What esp-module are you using? What kind of power supply?

Naesstrom, 2018-11-01:
NodeMCU v3 (lolin) and just USB power from the PC atm.

Naesstrom, 2018-11-03:
Running it with a 1A external poweradapter gives me:
```
Welcome to EspSparsnasGateway
Compiled at:Oct 30 2018 21:51:42
Debug on
Vcc=2923
Attempting MQTT connection...Connected to Mqtt broker as EspSparsnasGateway
Stored data: 
There are no stored frequency, using default value
There are no stored senderid, using default value
Senderid: 570631
Frequency: 868000000
61.04
Over The Air programming enabled, port: EspSparsnasGateway
Ready
IP address: 192.168.1.201
EspSparsnasGateway @ IP:192.168.1.201 SSID: Katrinelund
In initialize, frequency = 868100000
Adjusted freq: 14222950
61.04
Failed on waiting for ModeReady()
Unable to initialize the radio. Exiting.
```
Not sure what it means with `Failed on waiting for ModeReady()`

bphermansson, 2018-11-03:
Can you try to set the frequency in the code?

Naesstrom, 2018-11-07:
Redid the wiring on a breadboard but still the same result when connecting 12V power... Radio is getting 3.3V to.
```
Compiled at:Oct 30 2018 21:51:42
Debug on
Vcc=2928
Attempting MQTT connection...Connected to Mqtt broker as EspSparsnasGateway
Stored data: 
There are no stored frequency, using default value
There are no stored senderid, using default value
Senderid: 570631
Frequency: 868000000
61.04
Over The Air programming enabled, port: EspSparsnasGateway
Ready
IP address: 192.168.1.201
EspSparsnasGateway @ IP:192.168.1.201 SSID: Katrinelund
In initialize, frequency = 868100000
Adjusted freq: 14222950
61.04
Failed on waiting for ModeReady()
Unable to initialize the radio. Exiting.
```

Naesstrom, 2018-11-07:
Been googling and trying to find what might be causing it... only thing I found is related to the [rst pin](https://forum.mysensors.org/topic/5954/rfm69-not-initialising-required-reset-could-be-the-cause/5) but since it's been working for you it can't be that!

bphermansson, 2018-11-08:
Maybe try another software to see if it's a hardware or software problem? Like this: https://github.com/bbx10/nanohab

Naesstrom, 2018-11-08:
yup, that seems to be initializing the radio
```
RFM69 WiFi Gateway
External System
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.1.201
connected
*WM: freeing allocated params!
MDNS responder started
Connect to http://rfm69gw1.local or http://192.168.1.201

Listening at 868 86 MHz
Attempting MQTT connection...failed, rc=-2 try again in 2 seconds
```
Having some problems with my mqtt atm. tough but thats unrelated.

Just a thought, I'm using a RFM69HCW and in the software you linked he differs between them for some reason with `    radio.setHighPower();    // Only for RFM69HCW & HW!` can that be something?

Sorry for being such a PITA btw

bphermansson, 2018-11-15:
Might be, try to comment that line?

Naesstrom, 2018-11-19:
Not sure I follow what you mean right now, this software that you linked https://github.com/bbx10/nanohab managed to initialize the radio- The part with `radio.setHighPower` is from that git and not yours.
Do you mean I should comment that out from the above mentioned firmware and see if it still initializes the radio?

Naesstrom, 2018-12-10:
After switching mosi/miso I got it booting up atleast... D6 --> MOSI and D7 --> MISO
Can't really understand why since according to all the sheets I can read it shouldn't work this way...
[image]

```
Compiled at:Nov  8 2018 16:43:01
Debug on
Vcc=2928
Attempting MQTT connection...Connected to Mqtt broker as EspSparsnasGateway
Stored data: 
There are no stored frequency, using default value
There are no stored senderid, using default value
Senderid: 570631
Frequency: 868000000
61.04
Over The Air programming enabled, port: EspSparsnasGateway
Ready
IP address: 192.168.1.201
EspSparsnasGateway @ IP:192.168.1.201 SSID: Katrinelund
In initialize, frequency = 868100000
Adjusted freq: 14222950
61.04
RFM69 init done
Radio initialized.
Listening on 1023999936hz. Done in setup.
```

Freland, 2018-12-10:
> After switching mosi/miso I got it booting up atleast... D6 --> MOSI and D7 --> MISO
> Can't really understand why since according to all the sheets I can read it shouldn't work this way...
> [image] [...]

Seems like it is exactly the same issue I had when I tried getting it up and running a while back ago (https://github.com/bphermansson/EspSparsnasGateway/issues/20 )

Naesstrom, 2018-12-10:
hehe, ja det var ditt förslag i [elektronikforumet](http://elektronikforumet.com/forum/viewtopic.php?f=2&t=85006&sid=12e36dbaacd2049680aebbe939121b39&start=390#p1397193) som gav mig iden att testa att switcha plats på dom!
Har du någonsin fått det att fungera @Freland

Naesstrom, 2018-12-11:
Efter att ha diskuterat på FB, granskat kopplingsscheman etc. har jag en ny fundering... Enligt sommarlov's skiss så skall ju NSS kopplas till GPIO15 och det skriver @bphermansson i sin schematic!

Kollar jag däremot på Adafruits egna inkopplingsschema för RFM69HCW så har dom ingen koppling till EN då den automatiskt är på men däremot från samma GPIO till CS
[image]

Naesstrom, 2018-12-11:
Typ såhär:
[image]

dogmatic69, 2019-01-16:
> RST - this is the Reset pin for the radio. It's pulled high by default. Pull down to ground to put it into reset

Seems that you should pull it down to reset, not permanently.

Naesstrom, 2019-01-17:
Hmm, then it should be done in the code then... I'm running mine now with gnd connected to it and it seems to be running fine

dogmatic69, 2019-01-23:
I'm going to go ahead and close this issue as the code fixes are now in and the other part seems to be a hookup issue. If you still got an error open a new ticket for it.

closed 2019-01-23 by dogmatic69