bphermansson/EspSparsnasGateway #24: Compilation errors with Arduino IDE 1.8.8

kalj, 2019-01-06:
I get error messages like this:

```
EspSparsnasGateway:349:17: error: 'uint32_t' was not declared in this scope
 bool initialize(uint32_t frequency) {
                 ^
EspSparsnasGateway:604:14: error: variable or field 'setMode' declared void
 void setMode(uint8_t newMode) {
              ^
EspSparsnasGateway:604:14: error: 'uint8_t' was not declared in this scope
RFM69functions:1:1: error: 'uint32_t' does not name a type
 uint32_t getFrequency() {
 ^
RFM69functions:16:1: error: 'uint8_t' does not name a type
 uint8_t readReg(uint8_t addr) {
 ^
RFM69functions:24:15: error: variable or field 'writeReg' declared void
 void writeReg(uint8_t addr, uint8_t value) {
               ^
RFM69functions:24:15: error: 'uint8_t' was not declared in this scope
RFM69functions:24:29: error: 'uint8_t' was not declared in this scope
 void writeReg(uint8_t addr, uint8_t value) {
                             ^
RFM69functions:82:1: error: 'uint16_t' does not name a type
 uint16_t crc16(volatile uint8_t *data, size_t n) {
 ^
callback:2:28: error: 'byte' has not been declared
 void callback(char* topic, byte* payload, unsigned int length) {
                            ^
EspSparsnasGateway:109:6: error: previous declaration of 'void setup()' with 'C++' linkage
 void setup() {
      ^
In file included from /home/kalle/code/Arduino/EspSparsnasGateway/EspSparsnasGateway.ino:47:0:
/home/kalle/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Arduino.h:221:16: error: conflicts with new declaration with 'C' linkage
 void setup(void);
                ^
EspSparsnasGateway:650:6: error: previous declaration of 'void loop()' with 'C++' linkage
 void loop() {
      ^
In file included from /home/kalle/code/Arduino/EspSparsnasGateway/EspSparsnasGateway.ino:47:0:
/home/kalle/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/Arduino.h:222:15: error: conflicts with new declaration with 'C' linkage
 void loop(void);
               ^
/home/kalle/code/Arduino/EspSparsnasGateway/EspSparsnasGateway.ino: In function 'void setup()':
EspSparsnasGateway:134:30: error: no matching function for call to 'PubSubClient::setCallback(void (&)(char*, int*, unsigned int))'
   client.setCallback(callback); // What to do when a Mqtt message arrives
                              ^
/home/kalle/code/Arduino/EspSparsnasGateway/EspSparsnasGateway.ino:134:30: note: candidate is:
In file included from /home/kalle/code/Arduino/EspSparsnasGateway/EspSparsnasGateway.ino:71:0:
/home/kalle/code/Arduino/libraries/PubSubClient/src/PubSubClient.h:121:18: note: PubSubClient& PubSubClient::setCallback(std::function<void(char*, unsigned char*, unsigned int)>)
    PubSubClient& setCallback(MQTT_CALLBACK_SIGNATURE);
                  ^
/home/kalle/code/Arduino/libraries/PubSubClient/src/PubSubClient.h:121:18: note:   no known conversion for argument 1 from 'void(char*, int*, unsigned int)' to 'std::function<void(char*, unsigned char*, unsigned int)>'
EspSparsnasGateway:319:24: error: 'initialize' cannot be used as a function
   if (!initialize(ifreq)) {
                        ^
EspSparsnasGateway:340:57: error: 'getFrequency' was not declared in this scope
     String temp = "Listening on " + String(getFrequency()) + "hz. Done in setup.";
                                                         ^
/home/kalle/code/Arduino/EspSparsnasGateway/EspSparsnasGateway.ino: In function 'bool initialize(uint32_t)':
EspSparsnasGateway:349:35: error: 'bool initialize(uint32_t)' redeclared as different kind of symbol
 bool initialize(uint32_t frequency) {
                                   ^
EspSparsnasGateway:349:6: error: previous declaration of 'bool initialize'
 bool initialize(uint32_t frequency) {
      ^
EspSparsnasGateway:397:34: error: 'writeReg' was not declared in this scope
     writeReg(REG_SYNCVALUE1, 0xAA);
                                  ^
EspSparsnasGateway:399:34: error: 'readReg' was not declared in this scope
   } while (readReg(REG_SYNCVALUE1) != 0xaa && millis() - start < timeout);
                                  ^
EspSparsnasGateway:402:34: error: 'writeReg' was not declared in this scope
     writeReg(REG_SYNCVALUE1, 0x55);
                                  ^
EspSparsnasGateway:407:40: error: 'writeReg' was not declared in this scope
     writeReg(CONFIG[i][0], CONFIG[i][1]);
                                        ^
EspSparsnasGateway:411:28: error: 'setMode' was not declared in this scope
   setMode(RF69_MODE_STANDBY);
                            ^
/home/kalle/code/Arduino/EspSparsnasGateway/EspSparsnasGateway.ino: In function 'void interruptHandler()':
EspSparsnasGateway:458:54: error: 'readReg' was not declared in this scope
   if (_mode == RF69_MODE_RX && (readReg(REG_IRQFLAGS2) & RF_IRQFLAGS2_PAYLOADREADY)) {
                                                      ^
EspSparsnasGateway:466:30: error: 'setMode' was not declared in this scope
     setMode(RF69_MODE_STANDBY);
                              ^
EspSparsnasGateway:479:38: error: 'crc16' was not declared in this scope
     uint16_t crc = crc16(TEMPDATA, 18);
                                      ^
/home/kalle/code/Arduino/EspSparsnasGateway/EspSparsnasGateway.ino: In function 'void setMode(uint8_t)':
EspSparsnasGateway:613:35: error: 'readReg' was not declared in this scope
   uint8_t val = readReg(REG_OPMODE);
                                   ^
EspSparsnasGateway:616:64: error: 'writeReg' was not declared in this scope
       writeReg(REG_OPMODE, (val & 0xE3) | RF_OPMODE_TRANSMITTER);
                                                                ^
/home/kalle/code/Arduino/EspSparsnasGateway/RFM69functions.ino: In function 'uint32_t getFrequency()':
RFM69functions:2:53: error: 'readReg' was not declared in this scope
   return RF69_FSTEP * (((uint32_t)readReg(REG_FRFMSB) << 16) + ((uint16_t)readReg(REG_FRFMID) << 8) + readReg(REG_FRFLSB));
                                                     ^
/home/kalle/code/Arduino/EspSparsnasGateway/RFM69functions.ino: In function 'void receiveBegin()':
RFM69functions:7:28: error: 'readReg' was not declared in this scope
   if (readReg(REG_IRQFLAGS2) & RF_IRQFLAGS2_PAYLOADREADY) {
                            ^
RFM69functions:10:68: error: 'writeReg' was not declared in this scope
     writeReg(REG_PACKETCONFIG2, (val & 0xFB) | RF_PACKET2_RXRESTART);
```

I had to downgrade to Arduino IDE 1.6.13

dogmatic69, 2019-01-16:
I am running 1.8.9 built locally and it can compile just fine. Did you try compile one of the more basic blink sketch or similar from the esp libs?

Naesstrom, 2019-01-21:
I tried this today and got the same errors with 1.8.8. Since @dogmatic69 got it working on 1.8.9 I'm guessing there's some breaking changes with 1.8.8 that's fixed in 1.8.9

per1234, 2019-01-21:
If you take a look at the preprocessed sketch output, you can see the problem:
```c++
#line 55 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\EspSparsnasGateway.ino"
int __get_adc_mode(void);
#line 109 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\EspSparsnasGateway.ino"
void setup();
#line 349 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\EspSparsnasGateway.ino"
bool initialize(uint32_t frequency);
#line 458 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\EspSparsnasGateway.ino"
void interruptHandler();
#line 612 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\EspSparsnasGateway.ino"
void setMode(uint8_t newMode);
#line 658 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\EspSparsnasGateway.ino"
void loop();
#line 686 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\EspSparsnasGateway.ino"
void reconnect();
#line 1 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\RFM69functions.ino"
uint32_t getFrequency();
#line 4 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\RFM69functions.ino"
void receiveBegin();
#line 16 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\RFM69functions.ino"
uint8_t readReg(uint8_t addr);
#line 24 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\RFM69functions.ino"
void writeReg(uint8_t addr, uint8_t value);
#line 32 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\RFM69functions.ino"
void select();
#line 38 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\RFM69functions.ino"
void unselect();
#line 64 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\RFM69functions.ino"
bool receiveDone();
#line 82 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\RFM69functions.ino"
uint16_t crc16(volatile uint8_t *data, size_t n);
#line 2 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\callback.ino"
void callback(char* topic, byte* payload, unsigned int length);
#line 38 "C:\\Users\\per\\Desktop\\EspSparsnasGateway\\EspSparsnasGateway.ino"
#define appname "EspSparsnasGateway"

const char compile_date[] = __DATE__ " " __TIME__;

// Sometimes you need to change how files are included:
// If the code doesnt compile, try to comment the row below and uncomment the next:
//#include <RFM69registers.h>
#include "RFM69registers.h"

#include <Arduino.h>
```
The Arduino IDE generates function prototypes for all functions in .ino files which don't already have a prototype. Usually the Arduino IDE does a good job of this but in some rare cases it inserts the prototypes in the wrong place. That is what has happened here. The `#include` directive for Arduino.h must come before the function prototypes but the prototypes are being inserted above it. This bug has already been fixed in the Hourly Build of the Arduino IDE so there is no need to report it to Arduino.

The best workaround to fix the problem for the existing IDE release(s) with the bug is to move the `#include` directives to the top of the sketch.

An alternative workaround is to add function prototypes for the problematic functions in the correct location in the sketch.

lindehoff, 2019-01-21:
I think you will find your answer here: https://github.com/bblanchon/ArduinoJson/issues/756
Seams to be a bug in the beta version of ArduinoJson use the stable version instead.

closed 2019-01-23 by dogmatic69