bphermansson/EspSparsnasGateway #28: Bad data

dogmatic69, 2019-01-19:
Now I have it running, but the data seems to be bad. I've tried to change the flag in the config mentioned in the readme but that did not help.

```
    if (TEMPDATA[0] != 0x11 || TEMPDATA[1] != (SENSOR_ID & 0xFF) || TEMPDATA[3] != 0x07 || rcv_sensor_id != SENSOR_ID) {
      Serial.println("Bad data");

      Serial.print("data_0: ");
      Serial.println(TEMPDATA[0]);

      Serial.print("data_1: ");
      Serial.println(TEMPDATA[1]);

      Serial.print("SENSOR_ID & 0xFF: ");
      Serial.println(SENSOR_ID & 0xFF);

      Serial.print("data_3: ");
      Serial.println(TEMPDATA[3]);

      Serial.print("rcv_sensor_id: ");
      Serial.println(rcv_sensor_id);

      Serial.print("SENSOR_ID: ");
      Serial.println(SENSOR_ID);
```

output:

```
08:51:27.321 -> Got rf data
08:51:27.321 -> Bad data
08:51:27.321 -> data_0: 238
08:51:27.321 -> data_1: 101
08:51:27.321 -> SENSOR_ID & 0xFF: 93
08:51:27.355 -> data_3: 196
08:51:27.355 -> rcv_sensor_id: 215117836
08:51:27.355 -> SENSOR_ID: 542045
08:51:33.333 -> Got rf data
08:51:33.333 -> Bad data
08:51:33.333 -> data_0: 85
08:51:33.333 -> data_1: 93
08:51:33.333 -> SENSOR_ID & 0xFF: 93
08:51:33.333 -> data_3: 69
08:51:33.333 -> rcv_sensor_id: 4157798087
08:51:33.333 -> SENSOR_ID: 542045
```

dogmatic69, 2019-01-19:
Got it working now by commenting out the hard coded value for the sensor before `initialize`. Not sure why that is even there when you are setting it above or via MQTT.

Seen some other posts here about removing it so maybe it should be removed totally...
```
  // ifreq = 868100000;
```

Naesstrom, 2019-01-21:
Thanks, I'll make sure to edit that out of the code!

closed 2019-01-21 by Naesstrom