bphermansson/EspSparsnasGateway #31: Set TX power for RFM69HW

Naesstrom, 2019-01-21:
So reading up on the [manual for the BOB](https://cdn-learn.adafruit.com/downloads/pdf/adafruit-rfm69hcw-and-rfm96-rfm95-rfm98-lora-packet-padio-breakouts.pdf) I found a couple of pieces that might be interesting here.

1. Define a rst pin
```
#if defined (__AVR_ATmega328P__) // UNO or Feather 328P w/wing
#define RFM69_INT 3 //
#define RFM69_CS 4 //
#define RFM69_RST 2 // "A"
#define LED 13
#endif
```
### Setup
We begin by setting up the serial console and hard-resetting the RFM69
```
void setup()
{
Serial.begin(115200);
//while (!Serial) { delay(1); } // wait until serial console is open, remove if not tethered to computer
pinMode(LED, OUTPUT);
pinMode(RFM69_RST, OUTPUT);
digitalWrite(RFM69_RST, LOW);
Serial.println("Feather RFM69 RX Test!");
Serial.println();
// manual reset
digitalWrite(RFM69_RST, HIGH);
delay(10);
digitalWrite(RFM69_RST, LOW);
delay(10);
```

2. Setting the transmit power needed for the **HCW** board
### Initializing Radio
For transmission power you can select from 14 to 20 dBi. Lower numbers use less power, but have less range. **The second argument to the function is whether it is an HCW type radio, with extra amplifier. This should always be set to true!**
```
if (!rf69.init()) {
Serial.println("RFM69 radio init failed");
while (1);
}
Serial.println("RFM69 radio init OK!");
// Defaults after init are 434.0MHz, modulation GFSK_Rb250Fd250, +13dbM (for low power module)
// No encryption
if (!rf69.setFrequency(RF69_FREQ)) {
Serial.println("setFrequency failed");
}
// If you are using a high power RF69 eg RFM69HW, you *must* set a Tx power with the
// ishighpowermodule flag set like this:
rf69.setTxPower(20, true); // range from 14-20 for power, 2nd arg must be true for 69HCW
// The encryption key has to be the same as the one in the server
uint8_t key[] = { 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08,
0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08};
rf69.setEncryptionKey(key);
```

And after that it's just the normal transmit code but I think we could use some of these changes for this one since it's the RFM69**HCW** that's listed in the materials section!

dogmatic69, 2019-01-21:
Does TX power have any relevance in this project? We are receiving data only, no need to transmit.

Naesstrom, 2019-01-21:
> Does TX power have any relevance in this project? We are receiving data only, no need to transmit.

Ah true, so then it's only the setup we need to add and assign rst to a free pin!

kalj, 2019-01-22:
Great! This is probably the more correct way of doing this, rather than the crude RST-to-GND hack.

kalj, 2019-01-22:
We need to make sure to not be too specific to the Adafruit BoB. I think many people are using the plain RFM69HCW module and it would be great to have everything somewhat general. As far as I understand, the whole RST issue was only experienced using the Adafruit BoB. Can we generalize the code and/or the wiring?

kalj, 2019-01-22:
> Hmm, the wiring will probably have to be 2 different drawings since the
> module doesn't have that one but the code could be the same i guess... No
> coder tough but the program shouldn't stop if nothing happens when enabling
> rst?

Yeah, two different wirings or an explanation of the main differences is
probably enough.

I can look into setting up something robust for the code.

dogmatic69, 2019-01-22:
I've done some quick tests and I am not convinced on the ground -> rst issue.

Without it connected radio does not start. 

```
21:53:12.445 -> Sensor id sub: 2731498642
21:53:12.445 -> In initialize, frequency = 867990000
21:53:12.445 -> Adjusted freq: 14221148
21:53:12.445 -> 61.04
21:53:12.478 -> ERROR: Failed setting syncvalue1 1st time
21:53:12.511 -> Unable to initialize the radio. Exiting.
```

So with the ground connected and started I disconnect ground and then it stops transmitting data.

```
21:51:43.760 -> Sending to mqtt: {"seq":57524,"watt":1009.419,"total":279.793,"pulse":"279793","battery":100,"rssi":"-81","power":"3652"}
21:51:43.760 -> wifi connected: True
21:51:43.760 -> shits good, lets go!
21:51:43.760 -> Data here????
21:51:44.297 -> client.loop
21:51:45.298 -> client.loop
21:51:46.298 -> client.loop
21:51:47.296 -> client.loop
21:51:47.362 -> Got rf data
21:51:48.296 -> client.loop
21:51:48.296 -> client is availale
21:51:48.296 -> len: 2
21:51:48.296 -> Time to MQTTPINGRESP
21:51:49.294 -> client.loop
21:51:50.294 -> client.loop
21:51:51.297 -> client.loop
21:51:52.300 -> client.loop
21:51:53.295 -> client.loop
21:51:54.294 -> client.loop
21:51:55.295 -> client.loop
21:51:56.294 -> client.loop
21:51:57.294 -> client.loop
21:51:58.295 -> client.loop
21:51:58.761 -> Got rf data
21:51:58.761 -> Memory usage: 43376
21:51:58.761 -> Sending to mqtt: {"seq":57525,"watt":1001.195,"total":279.797,"pulse":"279797","battery":100,"rssi":"-87","power":"3682"}
21:51:58.761 -> wifi connected: True
21:51:58.761 -> shits good, lets go!
21:51:58.761 -> Data here????
21:51:58.861 -> Got rf data
21:51:59.294 -> client.loop
21:52:00.273 -> client.loop
21:52:01.270 -> client.loop
21:52:02.267 -> client.loop
21:52:03.297 -> client.loop
21:52:04.298 -> client.loop
21:52:05.295 -> client.loop
21:52:06.267 -> client.loop
21:52:06.401 -> Got rf data
21:52:07.298 -> client.loop
21:52:08.296 -> client.loop
21:52:09.295 -> client.loop
21:52:10.294 -> client.loop
21:52:10.294 -> client is availale
21:52:10.294 -> len: 2
21:52:10.294 -> Time to MQTTPINGRESP
21:52:11.294 -> client.loop
21:52:12.294 -> client.loop
21:52:13.294 -> client.loop
21:52:13.761 -> Got rf data
21:52:13.761 -> Memory usage: 43376
21:52:13.761 -> Sending to mqtt: {"seq":57526,"watt":1006.663,"total":279.801,"pulse":"279801","battery":100,"rssi":"-76","power":"3662"}
21:52:13.761 -> wifi connected: True
21:52:13.761 -> shits good, lets go!
21:52:13.761 -> Data here????
21:52:14.294 -> client.loop
21:52:14.827 -> Got rf data
21:52:15.295 -> client.loop
21:52:16.294 -> client.loop
21:52:17.298 -> client.loop
21:52:18.297 -> client.loop
21:52:18.829 -> Got rf data
21:52:19.295 -> client.loop
21:52:20.294 -> client.loop
21:52:21.294 -> client.loop
21:52:22.294 -> client.loop
21:52:23.296 -> client.loop
21:52:24.294 -> client.loop
21:52:25.294 -> client.loop
21:52:25.560 -> Got rf data
21:52:26.293 -> client.loop
21:52:27.296 -> client.loop
21:52:28.167 -> Got rf data
21:52:28.300 -> client.loop
21:52:28.766 -> Got rf data
21:52:28.766 -> Memory usage: 43376
21:52:28.766 -> Sending to mqtt: {"seq":57527,"watt":1005.565,"total":279.805,"pulse":"279805","battery":100,"rssi":"-81","power":"3666"}
21:52:28.766 -> wifi connected: True
21:52:28.766 -> shits good, lets go!

// disconnected ground around here. Before here you can see a message every ~ 15 seconds.

21:52:28.766 -> Data here????
21:52:29.297 -> client.loop
21:52:30.297 -> client.loop
21:52:30.766 -> Got rf data
21:52:31.299 -> client.loop
21:52:32.295 -> client.loop
21:52:32.295 -> client is availale
21:52:32.295 -> len: 2
21:52:32.295 -> Time to MQTTPINGRESP
21:52:33.295 -> client.loop
21:52:34.295 -> client.loop
21:52:35.295 -> client.loop
21:52:36.298 -> client.loop
21:52:36.365 -> Got rf data
21:52:37.295 -> client.loop
21:52:38.295 -> client.loop
21:52:39.292 -> client.loop
21:52:40.289 -> client.loop
21:52:41.286 -> client.loop
21:52:42.288 -> client.loop
21:52:43.284 -> client.loop
21:52:44.281 -> client.loop
21:52:45.284 -> client.loop
21:52:46.284 -> client.loop
21:52:47.283 -> client.loop
21:52:48.286 -> client.loop
21:52:49.282 -> client.loop
21:52:50.284 -> client.loop
21:52:51.282 -> client.loop
21:52:52.279 -> client.loop
21:52:52.279 -> client is availale
21:52:52.279 -> len: 2
21:52:52.279 -> Time to MQTTPINGRESP
21:52:53.279 -> client.loop
21:52:54.278 -> client.loop
21:52:55.280 -> client.loop
21:52:56.279 -> client.loop
21:52:57.276 -> client.loop
21:52:58.272 -> client.loop
21:52:59.301 -> client.loop
21:53:00.300 -> client.loop
21:53:01.296 -> client.loop
21:53:02.293 -> client.loop
21:53:03.290 -> client.loop
21:53:04.288 -> client.loop
21:53:05.291 -> client.loop
21:53:06.287 -> client.loop
21:53:07.286 -> client.loop
21:53:08.288 -> client.loop
21:53:09.285 -> client.loop
21:53:10.283 -> client.loop
21:53:10.781 -> ;l d⸮⸮| ⸮$⸮|⸮d⸮c|ǃ⸮⸮{⸮c⸮b⸮⸮og⸮l'o⸮⸮⸮#x⸮⸮l{            // ======= RESET pushed. its nearly 45 seconds without a message.
```

bphermansson, 2019-11-27:
Open this again if it's still relevant.

closed 2019-11-27 by bphermansson