bphermansson/EspSparsnasGateway #39: Merging or starting over?

bphermansson, 2019-07-02:
Looking at the code now I think it's a mess... I see some interesting forks, but I wonder why most of you have removed the extra source files and put all code in EspSparsnasGateway.ino? 
I'm thinking of making a "v2.0", but I like splitting the code. Any good reasons why I shouldn't?

dogmatic69, 2019-07-10:
@bphermansson you are the one that wrote the code :joy: 

Have a look at the blame on the `EspSparsnasGateway.ino` file and you will see the overwhelming majority of code is yours. Only changes I can see are bug fixes unless you want to highlight some specifics rather than sweeping statements.

bphermansson, 2019-07-10:
True:) 
I mean like in the feature/cleanup fork. But nevermind, I've started work on a v2 in Platormio/Atom. The code will be much cleaner and split in several files.

dogmatic69, 2019-07-10:
I merged it because there is so much code that is not used at all. Once all in the same place I had plans to split it out like you say.

bphermansson, 2019-07-10:
Ok, that's a good thought.

closed 2019-09-14 by bphermansson