bphermansson/EspSparsnasGateway #41: Socket error

riksarchen, 2019-09-14:
MQTT connects and then disconnects within a second. 

>1568413360: New client connected from 192.168.1.251 as EspSparsnasGatewayV2 (p2, c1, k30, u'hidden').
1568413360: Socket error on client EspSparsnasGatewayV2, disconnecting.
>
Worked fine with old version.

Hardware i use is a nodemcu esp8266, same as the one described in the guide.

bphermansson, 2019-09-14:
Tried to checkout and use the v1_original branch?
Do you use Mosquitto as broker software? If you do, try running it from the command line with the -d parameter.

riksarchen, 2019-09-15:
Tried v1. It does not seems to even try MQTT connection. I also tried -d and -v to see log. 

V1 does not connect at all, v2 connect and CONNACK is sent but then it's disconnected with Socket error. 

I'll try to swap the Amico ESP8266 to a wemos d1 mini.

riksarchen, 2019-09-28:
Tried with wemos D1 mini also now but still gets socket error as soon as it connects.

bphermansson, 2019-10-02:
Got this too now, with a circuit that's been working good before. Calls for further investigation. 

```
sudo cat /var/log/mosquitto/mosquitto.log
1570009721: New connection from 192.168.1.235 on port 1883.
1570009721: New client connected from 192.168.1.235 as mosqsub|18035-PatrikAth (c1, k60, u'emonpi').
1570009785: Socket error on client mosqsub|18035-PatrikAth, disconnecting.
```

bphermansson, 2019-11-27:
@riksarchen, you don't have this problem with your pubsub-version?