bphermansson/EspSparsnasGateway #48: Drastically wrong power reported when there is few pulses

nbasse, 2019-11-24:
Wrong power is being reported when there is a low load in the system making it take longer time between the light pulses on the meter.
For example, when there is 250-350W being consumed, the EspSparsnasGateway reports 10-20W, (with 100 pulses/Kwh for the meter, therefore approximately 3 minutes between the pulses).
But higher loads, for example, 2000w gets reported correctly.

Attached serial log shows first an apartment at approx 300w, then a kettle is turned on raising the power to above 2000w and then it is turned off again. 
(Most of the bad packets received have been filtered away to make the log shorter.)

[debug_log.txt](https://github.com/bphermansson/EspSparsnasGateway/files/3883963/debug_log.txt)

nbasse, 2019-11-24:
After performing more experiments, I have found the breaking-point to be somewhere between 599W (still working) and 557W (not working)

(All "real" values are from the original Sparsnäs display)

riksarchen, 2019-11-24:
I think the problem is not the decoded power but the number of pulses within a timespan. 

Your setup was 100 pulses per kWh, Confirm?

nbasse, 2019-11-24:
Yes, that is correct. 100 pulses /KWh

bphermansson, 2019-11-27:
I've added a function to print raw power values to the serial port. Not tested.

nbasse, 2019-11-28:
> I've added a function to print raw power values to the serial port. Not tested.

Not sure how I can test this. I am currently using @riksarchen fork as it has some tweaks to make it work with the board he deigned.

Some sort of raw values are already being printed, (see the log in my first post) are these different from the ones you added?

fredrike, 2020-03-26:
Could this be the same issue: https://github.com/tubalainen/sparsnas_decoder/pull/12/files

fredrike, 2020-03-28:
@nbasse, we need to see your `data4` value for the low power mode, but I suspect that we might have a case of `data4 = 3` which is not covered in the code..

https://github.com/bphermansson/EspSparsnasGateway/blob/38d3c45e60781a739a027dc910163e9dcb6d83e7/src/RFM69functions.cpp#L403-L410

nbasse, 2020-03-30:
@fredrike 
Data4 seems to be 1 at higher loads when everything is working and 0 when at lower loads when wrong power is being reported.

`Seq 18017: 1585594856, 624.72 W, total: 1190 kWh, battery 100%, rssi -108dBm. Power(raw): 59009 Vcc: 2993.00mV, Data4: 1`

`Seq 17972: 1585594181, 17.95 W, total: 1190 kWh, battery 100%, rssi -123dBm. Power(raw): 7478 Vcc: 2993.00mV, Data4: 0`

fredrike, 2020-03-31:
> Data4 seems to be 1 at higher loads when everything is working and 0 when at lower loads when wrong power is being reported.

Great! What is the consumtion with `Data4 = 0` and `Power(raw) = 7478` ? I.e., what formula should we use for that case?

nbasse, 2020-03-31:
Yes, started doing this yesterday, but only had the time to get 5 or 6 value pairs. And since the original display only shows whole watts, I think some more values would be good to get a better formula.

I started by fitting a straight line to the values. but I'm not sure where I should add the parameter for pluses/KWh. I will look into this more later today or tomorrow and try to get some more readings.

fredrike, 2020-03-31:
> I started by fitting a straight line to the values. but I'm not sure where I should add the parameter for pluses/KWh. I will look into this more later today or tomorrow and try to get some more readings.

Just post the readings here and I can try to find the factor for you.

nbasse, 2020-04-01:
OK, Here are some readings!
watt =[ 508 466 456 460 465 465 468 200 171 150 141 133 120 114 184 268 263 291 353]
raw = [4530 4934 5042 5003 4946 4948 4913 11501 13421 15341 16301 17261 19181 20055 12471 8575 8730 7895 6512]

Apparently it is not a linear relationship between the value pairs as I expected.
See attached plot
[image]

nbasse, 2020-08-01:
I have now fitted a exponential model to the values I recorded and extracted some coefficients.
I have successfully added it to the code and get good values even when data4 = 0. See https://github.com/nbasse/EspSparsnasGateway/commit/9f872e2a968904546745ac9a8cc33a35112ee8b3

I'm not sure if I correctly accounted for the number of pulses. So someone with a value other than 100 should try this to see if it is correct.
(Or maybe someone can just double-check the math? I just multiplied the value I got for C_a with 100 and then divided the calculated power with the number of pulses. Not sure if this is correct for all cases.)

fredrike, 2020-08-02:
Great job! 

I suggest that you send this as a pull-request so @bphermansson can merge it.