bphermansson/EspSparsnasGateway #50: Enable second transmitter id

SweNilsson, 2019-12-15:
By monitoring more than one energy meter at home it would be possible to keep consumption from the Car and the Heating system on separate "tabs". I imagine that the code change would be rather small, but I can't achieve this myself with my lacking programming skills. My current need is to receive data from two transmitters.

My idea to solve the need:
Setup transmitter_id2 in settings.h
either add watt2, total2 and battery2 to the MQTT message or create a separate message in case two transmitter ids have been set.

Sperra, 2020-02-13:
You can try this one: https://github.com/Sperra/EspSparsnasGateway

SweNilsson, 2020-02-14:
Thanks @Sperra ! What is the json value ids? I've tried to check the code but I'm not qualified to read it. Today I have a sensor reading this "value_json.watt" but now I need two different I guess?

Sperra, 2020-02-14:
During startup and initializing it uses the following for information and debug.
EspSparsnasGateway/values
EspSparsnasGateway/debug

Meassurment are sent with the following
EspSparsnasGateway/Sensor1/values
EspSparsnasGateway/Sensor1/debug
EspSparsnasGateway/Sensor2/values
EspSparsnasGateway/Sensor2/debug

Values are the same as before: seq, timestamp, watt, total, battery and rssi

I have the following setup for wat in my OpenHAB
Number sensors_sparsnas_watt "watt [%.1f W]" (gpower) {mqtt="<[mosquitto:EspSparsnasGateway/Sensor1/values:state:JSONPATH($.watt)]"}

SweNilsson, 2020-02-14:
Great! Exactly what I was looking for

closed 2020-02-14 by SweNilsson

bphermansson, 2020-02-14:
Great work! Just wished it'd been a PR here instead of a fork.

Sperra, 2020-02-15:
Sprry my faiult! Totaly new to GitHub. I will try to do a PR  during het weekend.

dogmatic69, 2020-02-15:
It's easy to make a [PR from a fork](https://github.com/bphermansson/EspSparsnasGateway/compare/master...Sperra:master)